elhep/GEM_HVSUP_eFUSE #4: Add DIP-switches or jumpers and standard resistors on mainboard

filipswit, 2025-02-03:
> - [ ] Add DIP-switches or jumpers and standard resistors on mainboard
> 
> 
> 
>  

 _Originally posted by @gkasprow in [#1](https://github.com/elhep/GEM_HVSUP_eFUSE/issues/1#issuecomment-1916560945)_

For what purpose?

gkasprow, 2025-02-04:
this is to adjust divider voltages. For the moment divider ratio is fixed and regulation is only within 50V or so.
It would be nice do add 4-channel DIP switch that can adjust it in wider range by shorting R, 2R, 4R and 8R segments

gkasprow, 2025-02-07:
this is a new divider circuit; the resistors are on the main board, and the rest is in the modules

[image]

filipswit, 2025-03-01:
Added resistors with 2.54mm jumpers

[image]