elhep/GEM_HVSUP_eFUSE #14: AVD current measurement

gkasprow, 2025-02-07:
> - [ ] remove ref diode; > 
> - [ ] reverse Zener diode> 
> - [ ] Rg=10k> 
> - [ ] add R shunt = 5k6/3 

 _Originally posted by @gkasprow in [#1](https://github.com/elhep/GEM_HVSUP_eFUSE/issues/1#issuecomment-1961682124)_

gkasprow, 2025-02-07:
nowe we have current measurement based on Zener, reference Zener 

[image]
and diff amplifier

[image]

that doesn't work well.

We are interested in th ecurrent difference between PNP and NPN parts

[image]
so  let's treat Zeners like overvoltage protection; add  1k8 in series and route both singnals to diff amp; get rid of R87;
R83 need to be connected to 1.2xV precision reference (some low cost reference diode like TS3431)

filipswit, 2025-03-01:
Is this what you were thinking?

[image]

[image]

gkasprow, 2025-03-02:
make it easier; imeas P, imeas N route to inverting opamps

[image]
we will calculate difference in digital way
you will need one extra ADC, you can use spare channels for 5V and 3.3V monitoring

filipswit, 2025-03-05:
I can add another ADC, but does it make sens to measure P12V0 with INA and ADC? When  we remove P12V0 measurement from ADC there will be room for AVD and MeasP/N

[image]

[image]

gkasprow, 2025-03-05:
Sure lets do that