elhep/GEM_HVSUP_eFUSE #16: BoM additionals

filipswit, 2025-02-09:
Here are PN propositions: 

`Add USB-A connector to power RPI`

17-200001
https://www.mouser.pl/ProductDetail/Amphenol-CONEC/17-200001?qs=5JGdZpAEOnhRoCbSSP33Ng%3D%3D

` Add USBA to micro USB cable to the BOM`

CA-USBM-10CC-0003-BK 0.3m Cable
https://www.morele.net/kabel-usb-lanberg-usb-a-microusb-0-3-m-czarny-ca-usbm-10cc-0003-bk-1620097/

` Add Panel Mount Ethernet Extension Cable to the BOM or CP30220SX Keystone mount`

RCP-5SPFFP-SCM7B10 it is with 100mm cable
https://www.mouser.pl/ProductDetail/Amphenol-LTW/RCP-5SPFFP-SCM7B10?qs=TFawuGRVzFrwnQdKv5DQbA%3D%3D

`add 15mm plastic standoffs to the BOM`

Is it for Raspberry Pi? Because then I think we need 30mm standoffs: HV module (19,6mm) + socket (8,5mm)

`add M3 6-8mm countersunk screws to the BOM`

How many?
https://www.tme.eu/pl/details/b3x6_bn388/sruby/bossard/1156454/

`add M3 8 mm Nylon screws to the BOM`

Is Poliamid fine?
https://www.tme.eu/pl/details/dr8g201m3x8/sruby/dremec/8g201m3x08/

`add M2.5 countersunk screws + nuts to the BOM (Ethernet adapter fixing)`

https://www.tme.eu/pl/details/b2.5x6_bn388/sruby/bossard/1156381/
https://www.tme.eu/pl/details/b2.5_bn109/nakretki/bossard/1088254/

` add rpi4 + SD card to the BOM`

Is 16G fine? for ex. PSF16GMDC10

`Add ethernet and usbc cables to BOM`
 
C1809GBK
C14103BK-1.5M

`Add hammond enclosure to BOM`

It already on BoM - HAMMOND 1550J

gkasprow, 2025-02-10:
> Add USB-A connector to power RPI

I meant PCB vertical USB-A connector; Very short USB cable will be used to power RPI installed above the main board

>  Add USBA to micro USB cable to the BOM
CA-USBM-10CC-0003-BK 0.3m Cable
https://www.morele.net/kabel-usb-lanberg-usb-a-microusb-0-3-m-czarny-ca-usbm-10cc-0003-bk-1620097/

OK

>Add Panel Mount Ethernet Extension Cable to the BOM or CP30220SX Keystone mount
RCP-5SPFFP-SCM7B10 it is with 100mm cable
https://www.mouser.pl/ProductDetail/Amphenol-LTW/RCP-5SPFFP-SCM7B10?qs=TFawuGRVzFrwnQdKv5DQbA%3D%3D

Use Keystone I mentioned, this one looks ugly  - the black thread will protrude through the box wall
https://www.tme.eu/pl/details/cp30220sx/zlacza-rj/cliff/

>add 15mm plastic standoffs to the BOM
Is it for Raspberry Pi? Because then I think we need 30mm standoffs: HV module (19,6mm) + socket (8,5mm)

Nope, for the main board - we need to attach it to the metal case
We need higher ones for RPI

>add M3 6-8mm countersunk screws to the BOM
How many?
https://www.tme.eu/pl/details/b3x6_bn388/sruby/bossard/1156454/

the same number as the standoffs above. We will use them to attach the PCB from bottom side of the metal box

>add M3 8 mm Nylon screws to the BOM
Is Poliamid fine?
https://www.tme.eu/pl/details/dr8g201m3x8/sruby/dremec/8g201m3x08/

yes, we will use them to attach the mainboard to the standoffs

>add M2.5 countersunk screws + nuts to the BOM (Ethernet adapter fixing)
https://www.tme.eu/pl/details/b2.5x6_bn388/sruby/bossard/1156381/
https://www.tme.eu/pl/details/b2.5_bn109/nakretki/bossard/1088254/

this is for ethernet connector (https://www.tme.eu/pl/details/cp30220sx/zlacza-rj/cliff/)

> add rpi4 + SD card to the BOM
Is 16G fine? for ex. PSF16GMDC10

no, 128GB high speed one (cat10); we have a lot of data to log

>Add ethernet and usbc cables to BOM
C1809GBK
C14103BK-1.5M

we need very short cable that connects RPI with Keystone connector

filipswit, 2025-02-23:
> I meant PCB vertical USB-A connector; Very short USB cable will be used to power RPI installed above the main board

Vertical USB connector [1734366-1](https://www.mouser.pl/ProductDetail/TE-Connectivity/1734366-1?qs=Y377PMEofyC1I8%2FZakOHrA%3D%3D)

15mm Standoffs for MainBoard [R30-1611500](https://www.tme.eu/pl/details/r30-1611500/tuleje-dystansowe-plastikowe/harwin/)

Added [Kingston 128GB uSD card](https://allegro.pl/oferta/kingston-karta-microsd-128gb-micro-cl10-adapter-sd-13956449208?bi_s=ads&bi_m=listing:desktop:product&bi_c=ZjczYmVhYmMtZGVmZS00ZDQ1LWExMDEtZDcxZDA5YWE0ZDU3AA&bi_t=ape&referrer=proxy&emission_unit_id=0f488b8d-d22e-4f93-9dbe-273041cb0c93)

Changed Eth cable to 0.15m [4040849924576](https://www.morele.net/goobay-cat-6-kabel-laczacy-s-ftp-pimf-pomaranczowy-dlugosc-kabla-0-15-m-10013485/)

Do you have a nut schematic symbol to use it? I have only screws and standoffs

[image]

gkasprow, 2025-02-23:
> Do you have a nut schematic symbol to use it? I have only screws and standoffs

just use general mechanic symbol I use in other projects; fill PN on the schematic

filipswit, 2025-03-15:
> Add USBA to micro USB cable to the BOM
> 
> CA-USBM-10CC-0003-BK 0.3m Cable
> https://www.morele.net/kabel-usb-lanberg-usb-a-microusb-0-3-m-czarny-ca-usbm-10cc-0003-bk-1620097/
> 

Rpi has USB-C connector, I swap it to short [USB-C cable](https://allegro.pl/oferta/boste-przewod-kabel-usb-a-usb-c-type-c-3a-quick-charge-mocny-szybki-25cm-15770899855)